gfroerli/hardware-v2 #1: Evaluate alternatives for oscillator

dbrgn, 2019-06-14:
Can/should we use a different one than on the previous board?

rnestler, 2019-06-15:
We should probably use the one recommended by the uC datasheet :wink:

dbrgn, 2019-11-29:
I did some more research in how to pick an oscillator and how to choose the capacitance.

I found these two links:

- [ST AN2867](https://www.st.com/content/ccc/resource/technical/document/application_note/c6/eb/5e/11/e3/69/43/eb/CD00221665.pdf/files/CD00221665.pdf/jcr:content/translations/en.CD00221665.pdf)
- https://blog.adafruit.com/2012/01/24/choosing-the-right-crystal-and-caps-for-your-design/

The oscillator we picked ([NX3215SA-32.768KHZ-EXS00A-MU00525](https://www.digikey.com/products/en?keywords=644-1386-1-ND)) has a load capacitance of 6 pF. Using the formula `(C*C) / (C+C) + Cs = 6` and a `C` value of 4.3 pF (the value we picked in the schematics right now), we end up with a stray capacitance of 3.85 pF to reach the target `6 pF`. However, the [datasheet](https://github.com/gfroerli/hardware-v2/blob/master/datasheets/stm32l071xxx.pdf) says that `10 pF` can be used as a rough estimate of Cs:

[image]

This text was about the HSE (high-speed) while the 32 kHz oscillator we use is a LSE. Does that guide still apply? Is this something we should test experimentally once we have a PCB?

rnestler, 2019-12-03:
Well we used the same values as the nucleo board, so I think we are fine.

dbrgn, 2019-12-07:
This can be closed, right?

commit e31bc5fe (2019-09-10): Use the oscillator design from the Nucleo board (#23)

Fixes #1
closed 2019-12-07 by dbrgn