gfroerli/hardware-v2 #12: Allow disconnecting RN from rest of the board

dbrgn, 2019-06-15:
This would allow flashing it without affecting the uC.

rnestler, 2019-09-10:
So there are two issues right?:
 * Short circuits  / conflicts on the UART pins
 * The 9V on the NRST pin when using the ICP

dbrgn, 2019-10-19:
> Short circuits / conflicts on the UART pins

Are there UART conflicts?

> The 9V on the NRST pin when using the ICP

Yep, that's the main issue.

rnestler, 2019-10-30:
> Are there UART conflicts?

Well yes: The PC sends stuff to update the FW and the uC sends stuff to send measurements over LoRa.

dbrgn, 2019-10-30:
Yeah, but during FW upgrade the uC should not be running at all, so that shouldn't be an issue.

rnestler, 2019-11-03:
> Yeah, but during FW upgrade the uC should not be running at all, so that shouldn't be an issue.

But then we need to take care to not powering the uC through the UART pins, which could lead to strange effects.

dbrgn, 2019-11-03:
Ok, good point. I wasn't aware that this could happen :)

rnestler, 2019-12-07:
> > The 9V on the NRST pin when using the ICP
>
> Yep, that's the main issue.

After thinking about it, the 10k resistor already protects the 3.3V rails from the 9V, but PA4 of the micro controller is also connected to it. So we'd need to protect that pin.

rnestler, 2019-12-07:
> Yeah, but during FW upgrade the uC should not be running at all, so that shouldn't be an issue.

We could pull the reset pin of the uC to low during FW upgrade of the RN module.

dbrgn, 2019-12-07:
That could work. It's not currently accessible through a header though, right?

dbrgn, 2019-12-10:
Power can be disconnected using the JP1 jumper.

If we need to reset the uC by pulling down STM_NRST, we can add a connection across C8.

closed 2019-12-10 by dbrgn