gfroerli/hardware-v2 #13: Connect RN2483 RST to uC

dbrgn, 2019-06-15:
Make it possible to reset RN2483 from microcontroller.

dbrgn, 2019-10-19:
Connected the RN NRST pin to PA4 of the uC in a106bb9cf53bd7193c020e8b031854147a7ee80d.

TODO:

- @rnestler can you confirm that this should work? (See commit message)
- We should test this using the nucleo devboard.

rnestler, 2019-12-07:
From the commit message:
> By default, after reset, the GPIO pins of the uC are configured as
floating inputs (see datasheet table 14). If we enable the pull-down
resistor on PA4, we should be able to reset the RN module.

Uhm what? If you have a pull up and a pull-down it will be a voltage in between the logic levels which may not work or even break stuff.
Why not do the obvious thing and configure it as an open drain output and set it to 0 to reset?

rnestler, 2019-12-07:
> We should test this using the nucleo devboard.

That's probably a smart thing to do anyways :)

dbrgn, 2019-12-07:
>  Why not do the obvious thing and configure it as an open drain output and set it to 0 to reset?

Uhh, that's what I meant to write :sweat_smile:

dbrgn, 2019-12-10:
We don't have the Nucleo board at hand, so we'll trust that it works :slightly_smiling_face:

closed 2019-12-10 by dbrgn