gfroerli/hardware-v2 #28: Check if we can measure the battery voltage below 3.3V

rnestler, 2019-12-03:
Currently it seems that we can only reliably measure the battery voltage down to 3.3V, because after that the ADC reference will also drop.

rnestler, 2019-12-03:
I guess we could use V<sub>REFINT</sub> for that.
[image]

dbrgn, 2019-12-07:
Hm, maybe I'm stupid, but I can't find the ADC_IN17 pin in the datasheet. (Other pins like ADC_IN10 or ADC_IN15 are listed in the pin definition table.

rnestler, 2019-12-07:
> ADC_IN17 pin

That's because it isn't a pin :wink: It is the *internal* bandgap which is connected internally to the `ADC_IN17` channel.

dbrgn, 2019-12-07:
Ah! So we wouldn't need to do any changes to the schematics to use this, right?

rnestler, 2019-12-10:
> Ah! So we wouldn't need to do any changes to the schematics to use this, right?

We'd probably need to change the voltage divider to keep the voltage between 0..V<sub>REFINT</sub>. But no change for the layout at least.

rnestler, 2020-08-04:
We could also consider using the V_LCD voltage monitoring:
[image]

dbrgn, 2021-07-20:
The internal ADC reference (V_REFINT):

[image]

Here are the specs:

[image]
[image]

dbrgn, 2021-07-20:
We need to adjust the voltage divider resistor values so that we can properly measure these values.

The exact reference value is stored in the `VREFINT_CAL` register. The max value is 1.242 V.

The resistor values must be chosen so that they map an input range of 0-5 V to the range 0-1.242V.

Existing divider:

[image]

We could use the following two values from the E24 1% series:

- R4: 9.1K
- R5: 3K

With this, we get:

- 1.242 V (MAX) for an input of 5.01 V
- 0.818 V for an input of 3.3 V
- 0.62 V for an input of 2.5 V

The ADC has 12 bit of accuracy. That means the ADC measures in steps of 0.3 mV (1.242 / 4096 * 1000). That should be plenty.

@rnestler does that look good? The ADC is saturating, so if we had a voltage of 5.1V we'd still get the max value (4096), right?

dbrgn, 2021-08-03:
(Right now, the measured voltage doesn't drop below 3.43 V.)