gfroerli/hardware-v2 #30: Connect RN2483 with antenna

dbrgn, 2019-12-04:
I gave it a try.

[Specs used by Aisler](https://aisler.net/help/design-rules-and-specifications/specifications):

- [FR-4-86](https://aisler.net/help/design-rules-and-specifications/specifications/2_layer_datasheet.pdf)
  - Dielectric constant / permittivity @ 1 MHz: Max 5.4 Measured 4.7
  - Loss Tangent @ 1 MHz: Max 0.035 Measured 0.02
- Copper thickness: 35 µm
- Plane clearance: 0.3mm

With 1.6mm board thickness:

[image]

With 2mm board thickness:

[image]

I assumed 1.6mm thickness and made an attempt at drawing the trace:

[image]

With the 1.415mm trace width we're at 50.0016 Ohms.

The length is roughly 17.44mm, this results in an angle(?) of 29.94°.

Does this look good? Are the input values correct? What does the angle mean?

dbrgn, 2019-12-04:
Interestingly, the [RN2483 datasheet](http://ww1.microchip.com/downloads/en/DeviceDoc/50002346C.pdf) suggests a different design:

[image]

dbrgn, 2019-12-04:
Note: The edge mount SMA connector is made for a board thickness of 1.57mm (space is 1.73mm).

rnestler, 2019-12-07:
> What does the angle mean?

Probably the resulting phase shift of the signal. Should increase proportional with the length of the trace.

rnestler, 2019-12-07:
In the old board we made it like this;
[image]
Since it worked quite well I guess we are fine as well.

Could you document all the decision making and sources for the PCB specs etc. in the `docs` folder?

dbrgn, 2019-12-07:
> Since it worked quite well I guess we are fine as well.

Ok! I'll also add a few more vias around the trace.

> Could you document all the decision making and sources for the PCB specs etc. in the docs folder?

Yep, can do that!

dbrgn, 2019-12-07:
> Probably the resulting phase shift of the signal. Should increase proportional with the length of the trace.

Yep, it does. Is it relevant for us?

rnestler, 2019-12-07:
> Plane clearance: 0.3mm

On https://aisler.net/help/design-rules-and-specifications/2-and-4-layer-design-rules it says that the min. spacing trace / ground plane is 0.2mm

dbrgn, 2019-12-07:
Ah, that part is not a spec by Aisler, it's the plane configuration we're using right now. I should have moved it to a different section, will fix.

dbrgn, 2019-12-09:
Ok, stitching vias added. Should be OK now.

closed 2019-12-09 by dbrgn

dbrgn, 2019-12-13:
Note: Footprint replaced with [this one](https://www.digikey.ch/products/en?keywords=%20CON-SMA-EDGE-S) due to price (1.68 CHF instead of >10).

[image]

The connector seems to be basically the same as the [Samtec SMA-J-P-H-ST-EM1](https://www.digikey.com/product-detail/en/samtec-inc/SMA-J-P-H-ST-EM1/SAM8857-ND/2602450).