gfroerli/hardware-v2 #31: SMA connector fixes

dbrgn, 2019-12-04:
[image]

Either remove those vias, or connect them to the GND net.

closed 2019-12-06 by dbrgn

rnestler, 2019-12-10:
Is this really fixed? It still shows up as a DRC error for me :/

dbrgn, 2019-12-10:
> Is this really fixed? It still shows up as a DRC error for me :/

Hm, you mean these errors?

[image]

Strange, clearance between "GND" and ""... Looks like a bug to me.

The device looks like this:

[image]

So the two pads are connected to the same net as the GND legs.