gfroerli/hardware-v2 #34: Replace LDO

dbrgn, 2021-05-13:
Right now we're using the [LP5907-Q1](https://www.ti.com/lit/ds/symlink/lp5907-q1.pdf?ts=1620922927380&ref_url=https%253A%252F%252Fwww.ti.com%252Fstore%252Fti%252Fen%252Fp%252Fproduct%252F%253Fp%253DLP5907QMFX-3.3Q1). I'm not sure what we were thinking when picking it (probably nothing), but it has an I_Q of at least 12 uA (in practice it seems to be around 15-16 uA).

These two models seem to be drop-in replacements:

- [TPS7A02 (25nA I_Q, 1.23 CHF @ Q10)](https://www.ti.com/lit/ds/symlink/tps7a02.pdf?ts=1620923734192&ref_url=https%253A%252F%252Fwww.ti.com%252Fassets%252Fjs%252FcompareParts%252Fcompare.html%253FfamilyId%253D400%2526parts%253DTPS7A02%252CTPS7A03%252CTPS7A20%2526cols%253Do1%252Cp182%252Cp1154%252Cp236typ%252Cp238min%252Cp451max%252Cp238max%252Cp634min%252Cp634max%252Cp1128%252Cp2954%252Cp2277%252Cp1130%252Cp3316%2526lang%253Den)
- [TPS7A03 (200nA I_Q, 0.73 CHF @ Q10)](https://www.ti.com/lit/ds/symlink/tps7a03.pdf?ts=1620896599862&ref_url=https%253A%252F%252Fwww.ti.com%252Fpower-management%252Flinear-regulators-ldo%252Fproducts.html)

They can deliver up to 200 mA (previous model offered 250 mA), but that should be sufficient.

To be determined:

- Will they work as drop-in replacements? (Pinout seems to be the same)
- Which one? (Price isn't really an issue, so I'd go with the 25nA TPS7A02 even if it's expensive.)

CC @rnestler

rnestler, 2021-05-19:
Well the dropout voltage seems higher (but difficult to compare, since they give it only for 200mA load).

dbrgn, 2021-05-19:
> since they give it only for 200mA load

From the datasheet:

[image]

dbrgn, 2021-05-19:
The one on the left is for 5V output voltage. We're using 3.3V, so it should be lower. Here's 1.8V:

[image]

rnestler, 2021-05-19:
Ah, I did only check the tables. The old one had:
[image]

rnestler, 2021-05-19:
> We're using 3.3V, so it should be lower. Here's 1.8V

Well if you look at the graph, at 50mA (which we need during TX) the dropout is around 100mA@1.8V while it is 80mA@5V. So I don't agree that it is lower.

dbrgn, 2021-05-19:
Ah, you're right, I missed that.

The main issue here is that the uC might brown-out due to the high dropout voltage when we have current spikes due to LoRaWAN transmissions, right?

rnestler, 2021-05-19:
> The main issue here is that the uC might brown-out due to the high dropout voltage when we have current spikes due to LoRaWAN transmissions, right?

Yes. But also less run time if the battery voltage goes down.

dbrgn, 2021-05-19:
Sure, but during sleep the power consumption is really low, so we're mostly bound by the peak current, right?

So, it seems to be an optimization problem bound by LDO quiescent current, LDO drop-out voltage, avg current during sleep, avg current during measurement, peak current during transmission and duration of the measurement phase.

Are there tools to calculate this optimization problem? I assume it should be pretty common?

rnestler, 2021-05-19:
IMO the quiescent current is more important for us, since our battery won't go below 3.3V until it is almost empty anyways, right?

dbrgn, 2021-05-19:
We'd need to profile that, but yes, I'd estimate that when it goes below 3.3V, less than 10% of the capacity are left.

See this chart, taken from [here](https://lygte-info.dk/review/batteries2012/Samsung%20INR18650-35E%203500mAh%20%28Pink%29%20UK.html):

[image]

(This is one of the batteries we're using.)

dbrgn, 2021-05-29:
For the record, we [compared different models](https://md.coredump.ch/gfroerli-ldo) and found two candidates:

- [Ablic S-1317A33-M5T1U4](https://www.digikey.ch/product-detail/de/ablic-u-s-a-inc/S-1317A33-M5T1U4/1662-2848-1-ND/7927985)
- [TI TPS782](https://www.digikey.ch/product-detail/de/texas-instruments/TPS78233DDCR/296-24059-1-ND/1990152)

We wanted to test and compare both, but unfortunately the TPS782 is not in stock on Digikey. Therefore we'll first try the Ablic LDO. It's pin-compatible with the current LDO.

Note: The TPS872 requires pin 4 to be connected to GND, while right now it's floating. On the prototype we can solder in a jumper wire, but we should fix this on the next revision: https://github.com/gfroerli/hardware-v2/issues/36

dbrgn, 2021-07-20:
The Ablic LDO works great!

Fixed in https://github.com/gfroerli/hardware-v2/commit/858049d78f3449aeb47347d26a1fda94862666c2.

closed 2021-07-20 by dbrgn